fixajteknik/YouTube_Tutorials #5: LoraE22 ve Esp32 kullanımı

burakturkers, 2024-12-26:
Merhabalar bitirme projem için lora modülü ile ses haberleşmesi yapmak için TS1912286 sipariş numaralı malzemeleri almıştım. Lakin macbook kullandığım için tam olarak esp32 ve lora modülünün arduino ile kodlamasını ve çalıştırmasını yapamadım acaba macbook için ayrı bir tanıtım videonuz var mıdır veya kodlamam için destek olabilir misiniz ?

fixajteknik, 2024-12-26:
aslında arduino ide de macbook için ayrı bir işleme gerek yok ama sizin için internette bulduğum ve beğendiğim bir kaç içerik var arduino başlangıç için onları buraya yazayım

https://www.youtube.com/watch?v=_2JJ29kmwRU

burakturkers, 2024-12-27:
Teşekkürler bu adımları gerçekleştirdim ve sorunumu çözdüm.