fixajteknik/YouTube_Tutorials #6: Module Selection for Most Accurate RSSI Data

tolgaSelimoglu, 2024-12-30:
According to your expreiences, which LoRa Module would you prefer in an application that RSSI data is important, 
E22, E32, E70 or otherwise?

fixajteknik, 2024-12-30:
rssi ölçümü e22 ve e70 modülleri ile yaplabilir. aynı şekilde rakbdk LoraWAN modülümüz ile de ölçüm yapılabilir. projenizin detayını anlatabilir misiniz

tolgaSelimoglu, 2024-12-30:
> rssi ölçümü e22 ve e70 modülleri ile yaplabilir. aynı şekilde rakbdk LoraWAN modülümüz ile de ölçüm yapılabilir. projenizin detayını anlatabilir misiniz

RSSI ölçümünü olabildiğince hızlı ve doğru bir şekilde almamız gerekli. Ne kadar doğru olursa eğitilecek olan neural network beklentimiz doğrultusunda daha doğru çalışacaktır. En hızlı iletişim ve en doğru RSSI ölçümü projenin isterlerinden. E22 veya E70 in bu konularda biribirine olan üstünlüğünü öğrenmek istiyorum aslında.

fixajteknik, 2025-01-06:
çok bi farkı yok rssi olarak. E70 deki mantık aynı anda birden fazla noktadan mesaj geldiğinde alabiliyor olması.