fixajteknik/YouTube_Tutorials #8: LoRa E22 900T22D Anten Sorusu.

neodevsoft, 2025-01-04:
### Şart CheckBox'ları işaretleyin.

- [X] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [X] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [X] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [X] Satın aldığım ürünlere yorum yaptım.
- [X] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

TS3012305

### Yorum yazdığınız kullanıcı adı nedir?

Null

### Arduino IDE veya IDF versiyonu.

Null

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

Uzman

### Eğitim seviyeniz

Üniversite

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

Merhaba, sitenizden aldığım LoRa E22 900T22D ile kullanabileceğim bu tarz anten elinizde mevcut mu ? ya da getirmeyi düşünüyor musunuz ? Yine sizden E22'ile kullanmak için "5.0dB 868 mhz 900 mhz 4g LTE 698~960MHz,1710~2700MHz SMA " anteni aldım. Fakat bu anten outdoor kullanımlar için uygun değil yapısı gereği kırılma ihtimalleri var. LoRa ile kullanmak için bu tarz esnek antenlere ihtiyacımız var. Teşekkürler.

[image]
[image]

### Ama gerçekte ne oldu?

Null

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

1. Adım
2. Adım
3. Adım
...

fixajteknik, 2025-01-06:
şu an stoklarımızda bu esnek olandan yok ama low SWR antednde robber antendir ona bakabilirsiniz. dış ortamda 6 yıldır kullandığım yer var sıkıntı yaşamadım.

https://shop.fixaj.com/dusuk-swr-li-915mhz-43-dbm-sma-anten

neodevsoft, 2025-01-06:
Cevabınız için teşekkür ederim, evet bu ürününüzüde inceledim hatta youtube videolarınızda da bahsediyorsunuz dış ortamda kullanmak için en ideal antenin bu olduğunu. Fakat projelerimde kullanmak için esnek bir antene ihtiyacım var. Eğer bu tür bir anten getirirseniz lütfen sosyal medya hesaplarınızdan bilgi verin çünkü bekliyorum. İyi çalışmalar.