fixajteknik/YouTube_Tutorials #9: Esp32-E22 Lora ile INMP441 mikrofon bağlantısı

burakturkers, 2025-01-06:
Merhabalar 3in1 ESP PCB ile E22Lora haberleşmesini sağladım ama yapıcağım projede ESP32 ile de INMP441 çok yönlü mikrofon bağlamam lazım ama 3in1 ESP PCB de giriş ve ayarlayabileceğim pinleri çözemedim çünkü boşta sadece Rx2 Tx2 pinlerimiz boşta bu konuda yardımcı olabilir misiniz?

fixajteknik, 2025-01-06:
S3 pcb mizde o mikrofon için hazır yer mevcut pin uclarını serbest yapmışım istediğiniz pine lehimleyebilirsiniz.
[image]

fixajteknik, 2025-01-06:
https://shop.fixaj.com/esp32-s3-board-n16r8-lehimlenmis-tak-calistir-usb-type-c

fixajteknik, 2025-01-06:
schematic de istersen şöyle mevcut s3 pcb mizde
[image]

burakturkers, 2025-01-06:
teşekkürler ama s3 pcb olmadan elimdeki 3in1 pcblerde çalıştırmam için bir bağlantı istemiştim aslında çünkü araştırmalarım sonucunda giriş pinlerini bulamadım

fixajteknik, 2025-01-06:
işin kolayı varken