fixajteknik/YouTube_Tutorials #12: Canbus Stm32

GaaraClood, 2025-01-15:
### Şart, TÜM CheckBox'ları işaretleyin.

- [X] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [X] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [X] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [X] Satın aldığım ürünlere yorum yaptım.
- [X] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

4582652

### Yorum yazdığınız kullanıcı adı nedir?

xzxgaaraxzx

### Arduino IDE veya IDF versiyonu.

Stm32 kubemix

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

SIFIR

### Eğitim seviyeniz

None

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

Hocam elimde 1 adet arduino uno ve buna bağlı olarak mcp 2515 var ve bir canbus snffing kodu ile python kodunda yazılmış bir Id ve 8 bitlik veri okuyucu program var ben motor siklet ecu suna bağlanıp bu can high ve can low bağlantılarını yapıp kontağı çevirdiğimde 

6 adet farklı Id ve 8 bit veri geliyor 

Burası Ok 

Fakat benim amacım bir stm32f103c8t6 ve 1 adet mcp2551 kullanarak bu karttan X Id de ve 8 bit ızunluğunda herhangi bir veri göndermek istiyorum fakat bunu beceremedim 😐.

### Ama gerçekte ne oldu?

Aslında 2 adet youtube videosunu birleşştirmeye çalışıyorum bu ve bu bvideo yu.
https://youtu.be/grcHzghNrKA?si=VoXuN8YaKyUvJIfi
https://www.youtube.com/watch?v=ZhYc95b6WoU&t=1181s

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

1. Adım
2. Adım
3. Adım
...

GaaraClood, 2025-01-15:
Şimdiden çok teşekkür ederim.

fixajteknik, 2025-01-16:
kendi sattığımız ürünlere destek veriyoruz