fixajteknik/YouTube_Tutorials #13: Rs232 haberleşme Teknofest Roket

Osmancidamal, 2025-01-16:
Merhaba,

Ben Osman Can, FlyIST Roket Takımı Aviyonik  üyesiyim. Bu yıl Teknofest Roket Yarışması'nda kullanılan uçuş bilgisayarıyla ilgili yeni eklenen bir şartla karşılaştık. Bu şart, kendi tasarladığımız uçuş bilgisayarına yerdeyken bir RS232 arayüzüne sahip test cihazı bağlayarak iki temel testi gerçekleştirmemizi gerektiriyor:

Sensörlerden alınan veriler üzerinden dalgalanma ve parazitlerin gözlemlenmesi.
Sanal olarak gönderilen açı, yükseklik gibi sensör verileriyle bilgisayarın paraşüt açma emrini hangi şartlarda verdiğinin gözlemlenmesi.
Bu testlerin gerçekleştirilmesi için uçuş bilgisayarımızın üç ayrı UART haberleşmesini eşzamanlı olarak yapması gerekiyor: GPS, LoRa ve RS232. Mevcut planımız şu şekildedir:

GPS haberleşmesi için SoftwareSerial kullanmak,
LoRa modülü için birinci HardwareSerial portunu kullanmak,
RS232 için ikinci HardwareSerial portunu tahsis etmek.
Bu düzenleme uygun mudur, ya da daha verimli bir çözüm öneriniz var mı? Ayrıca, bu konuda teknik açıklamalar ve öneriler içeren bir video hazırlarsanız, yarışmaya katılan tüm takımlar için çok faydalı olacağını düşünüyoruz.

Destekleriniz için şimdiden teşekkür ederiz.

Saygılarımızla,
FlyIST Roket Takımı

fixajteknik, 2025-01-16:
s3 pcb mizi incelediniz mi

Osmancidamal, 2025-01-16:
evet inceledim ve özelliklerini çok beğendik lakin bu pcb leri test pcb si olarak kullanıp dah sonra kendi pcb mize geçeceğiz. s3pcb gibi şematik olmadan tamamen smd yapıcak kadar imkan ve bilgimiz yok. ayrıca normal s3 lerde sizin de videonuzda bahsettiğiniz gibi uart 0 kullanılamıyor. bu yüzden  wroom üstünden bu ihtiyacı karşılamaya çalışıyoruz. Esp 32 nin rx0 ve tx0 pinlerinin yerlerini değitirip (çünkü aynı zamanda o pinler flash hafıza tarafından kullanılıyor.) bir dipswitch kullanarak uart0 ı kullanan modülün bağlantısını kesip açarak uart0 ı hem pc bağlantısı için hem de modül için kullanabilir miyim. Sizce bu bizim projemiz için uygun olur mu

fixajteknik, 2025-01-17:
ben orda esp32 s3 ün USB interface kullandım esp32 devkit usb desteği yok nasyıl yaparsınız bilmiyorum

Osmancidamal, 2025-01-18:
hocam bir dipswitch koysak hem modüllerin rx tx ini değiştiriyor hem de esp ye sinyal göndererek yazılımdan diğer modül seçiliyor çalışır mı

fixajteknik, 2025-01-20:
Bilmiyorum denemedim

ankaralii, 2025-03-29:
hocam selamlar, rs232 bir nevi uart protokolü. veri göndermenizde software serial yapmak ya da hardware serial yapmaktan daha önemli olan unsur kaç tane uart portunun olduğu. eğer sen, sensörlerin+lora+rs232 çıkışını karşılayacak kadar uart portuna sahipsen hangisinin software hangisinin hardware olduğunun önemi yok. lora için parametre ayarlarını doğru yap, geriye kalanları da uart olarak gönder ya da al, zaten çözüm olur. 

ötr'yi geçtiğiniz için tebrik ederim. bir gömülü yazılımcıyla bunları konuşursanız işleriniz rahatlar. iyi çalışmalar

Osmancidamal, 2025-03-29:
teşekkür ederim mega pro mini ye geçerek sorunu çözdük