fixajteknik/YouTube_Tutorials #14: LoraWAN RAK3172H BDK

Mehmetyr03, 2025-01-17:
Merhaba, Sirapiş no: TS0312257,
rak bdk pinout pinlerinden UART2  pinlerini kullanarak rak3172 nin kendisini programlamak istiyorum, bağlantımı USB-UART çevirici ve ESP32 cihazlarlarının ikisi ilede denememe rağmen Boot moda giremiyorum ve rak3172 ile haberleşmeye geçemiyorum. Lorawan videonuzu ve github'daki örneklerinizi inceledim ama bir yol bulamadım. STM32CubeProgrammer ve Ardinuo IDE ikisinide kullanmaktayım. 
STM32CubeProgrammer; (read Unprotecect seçeneği işaretli)
 -Warning: Device is under Read Out Protection
 -Device Read Unprotect requested
 -Response received from device: NACK
 -Error: GETID command not acknowledged!
 -Error: READ command not acknowledged at address: 0x1FFF73F8
bağlantısı okumaya karşı korumalı olduğu uyarısı vermektedir.  

Ardinuo IDE örnek bir kod yüklemeye çalıştığım zaman ;
device is not in boot mode
Detecting baudrate..................
Detect baudrate fail, can not get the baudrate

Sormak istediğim rak3172 yi nasıl boot moda alabilirim?

fixajteknik, 2025-01-17:
anladım benim firmware değilde sen kendin programın ile rak modülünü kullanmak istiyorsun aslında çok kolay.

bana bağlantı şemanın resimlerini atar mısın. nasıl programladığını görmek istiyorum. varsa video da olur.

ayrıca bu hata değildir ```
Detecting baudrate..................
Detect baudrate fail, can not get the baudrate
```

bundan sonra biraz bekleyince program atabilirsiniz. şimdilik ekran görüntüleri ve nasıl hata aldığını görebileceğim görseller gerekli

fixajteknik, 2025-01-17:
Ne yaptin hocam deneye bildin mi yapamadiysan kisa bir video cekeyim

Mehmetyr03, 2025-01-19:
Merhaba, işi doktoruna sorunca çözüm yolu bulundu. Boot moda sokmaya ve rak3172 ile At komutları ile iletişime engel olan Boot0 pininin high modda olması imiş. Hemen hemen bütün kaynak ve talimatlarıda boot0 pini high olması şeklinde olduğu idi. Boot0 pini durumunu değiştirince at komutları çalıştı ve yeni programı yükleyebildim. teşekkürler

fixajteknik, 2025-01-20:
Videosunu da çektim bakarsınız

https://www.youtube.com/watch?v=OiPGeWgaJ60

Mehmetyr03, 2025-02-03:
Merhaba,  program yüklemsi yaptığım ve at komutları gönderebildiğim rak bdk çalışıyordu. Binin üzerinde lorawan haberleşmesi yaptı ve çaışırken bir anda sorun oluştu, seriport ekranında anlamsız sürekli ilerleyen bir şekilde semboller çıkmaya başladı,  Artık hiçbir şekilde at komutları gönderemiyor ve yeniden programlayamıyorum.

fixajteknik, 2025-02-03:
şu video da anlattığım gibi st link ile flash la yeniden resetle düzelir.

Mehmetyr03, 2025-02-03:
st link mevcut değil malesef, başka bir yolu var mı acaba?

fixajteknik, 2025-02-03:
ST link temin etmelisiniz çünkü ileride de size lazım olacak ST link üzerinden flashlamak en kötü seneryoda bile işe yarar

Mehmetyr03, 2025-02-03:
Teşekkürler

closed 2025-08-14 by fixajteknik