fixajteknik/YouTube_Tutorials #19: Lora RF Settings Hatası

Valeron1853, 2025-02-01:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

7382657

### Yorum yazdığınız kullanıcı adı nedir?

Salih ÇİL

### Arduino IDE veya IDF versiyonu.

Arduino IDE 2.3.4

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

1 Yıldan Az

### Eğitim seviyeniz

Üniversite

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

Açıkçası başlangıç aşamasındayım hocam. Öncelikle başka sitelerden E22400T30D modelini frekansa uygun anteni ve arduino uno kartlarımı aldım. Kodları yazdığımda RF settings ayarlarını başarıyla yaptım. Fakat msj gönderirken konfigürasyon hatası aldım. Daha sonra Lora larda arıza olabileceğini düşünerek USB stik aldım. Loralar çalışıyordu. Ayar modunda Serial Monitörde AT komutunu girdiğimde üçgen içinde ??? adet soru işareti alıyordum.

Bu sefer arduinolarda sorun olabileceğini düşündüm. 
Sitenizde Arduino Mini hediyeli PCB kart aldım 2 adet. Bu sefer de denedim aynı şekilde. RF Settings çalıştı. Ama yine aynı hataları aldım hem konfigürasyon hatası hem de AT komutunu girdiğimde ??? dönüşü oluyor.

Yardımcı olabilir misiniz?

### Ama gerçekte ne oldu?

#include <SoftwareSerial.h>
   
SoftwareSerial mySerial(3, 4); // RX, TX
   
#define M0 7
#define M1 6
   
void setup() {
  Serial.begin(9600);
  mySerial.begin(9600);
   
  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  digitalWrite(M0, LOW);
  digitalWrite(M1, HIGH);
}
   
void loop() {
  if (Serial.available()) {
    mySerial.write(Serial.read());
  }
   
  if (mySerial.available()) {
    Serial.write(mySerial.read());
  }
}

[image]

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

10:56:01.233 -> ���

[image]

### Log çıktıları varsa ekleyin.

```plain

```

Valeron1853, 2025-02-01:
[image]

Valeron1853, 2025-02-01:
[image]

Valeron1853, 2025-02-01:
Hocam https://fixaj.com/kolay-parametre-ayarlari/ kullanarak modüller arası iletişimi sağladım.
Sorun yok gözüküyor şimdilik. 

Fakat bu RF settings modda serial monitörde  doğru baud ayarında ve BOTH NL & CR seçeneğinde modül'ün bize "AT" yazınca "OK" olarak dönüş vermiyor. ??? işareti çıkıyor, o kısmı çözemedim, bilmiyorum buraya takılmam normal mi

fixajteknik, 2025-02-01:
Modulu satmadigim icin bilgim yok maalesef

closed 2025-08-15 by fixajteknik