fixajteknik/YouTube_Tutorials #20: rf setting ayar proramı çalışmadı

vedat504, 2025-02-03:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

sipariş telefon üstünden oldo

### Yorum yazdığınız kullanıcı adı nedir?

vedat504

### Arduino IDE veya IDF versiyonu.

IDE 2.3.4

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

SIFIR

### Eğitim seviyeniz

Lise

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

[image]

### Ama gerçekte ne oldu?

Hayırlı İşleriz olsun
indirme sayfasından ardunio proramını indirip kuruyorum. E22 için M0 aktif edin yazıyor ve M0 HIGH yapıyorum, rf setting programi çalışmıyor.
programda değişiklik yapmadan yükleme yaparsam RF setting çalışıyor.
loralar arası veri iletimi yapamadım. röle açıp kapama yapacağım ama başaramadım. yardımcı olursanız sevinirim.

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

1. Adım
2. Adım
3. Adım
....

### Log çıktıları varsa ekleyin.

```plain
çalışmadı
```

### Daha fazla bilgi

çalışmadı

fixajteknik, 2025-02-03:
hocam bunlara gerek yok. şu video ve şu kodlara bakman yeterli parametre ayarlarını program dinamik olarak verir. hangi ayarın ne anlama geldiğini anlamak için ve nasıl çalıştırılacağını öğrenmek için videoyu dikkatlice izleyin:

VİDEO: https://www.youtube.com/watch?v=13nKOj_uG2s

KOD: https://github.com/fixajteknik/YouTube_Tutorials/tree/main/Video%20108%20%23define%20Part%202

vedat504, 2025-02-15:
Teşekkür ederim sorun çözüldü.