fixajteknik/YouTube_Tutorials #25: rf setting e22 modülünde bu işlemleri yapıyorum

dostt66, 2025-02-11:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

TS0602401

### Yorum yazdığınız kullanıcı adı nedir?

meraklo

### Arduino IDE veya IDF versiyonu.

ardinio 2,3,4

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

SIFIR

### Eğitim seviyeniz

Üniversite

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

rf setting e22 modülünde bu işlemleri yapıyorum  ama ekrandaki hatayı alıyorum[

### Ama gerçekte ne oldu?

rf setting e22 modülünde bu işlemleri yapıyorum  ama ekrandaki hatayı alıyorum

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

1. Adım
2. Adım
3. Adım
...

### Log çıktıları varsa ekleyin.

```plain

```

### Daha fazla bilgi

[image]
[image]

fixajteknik, 2025-02-11:
usb stick görselini de atar mısın boşta

dostt66, 2025-02-11:
[image]

fixajteknik, 2025-02-11:
usb stick yerine anladığım kadarı ile bizim 3in1 pcb kullanılıyor doğru mu?

dostt66, 2025-02-11:
evet

fixajteknik, 2025-02-11:
[RF_Setting(E22-E9X(SL)) V2.8.zip](https://github.com/user-attachments/files/18749011/RF_Setting.E22-E9X.SL.V2.8.zip)

dostt66, 2025-02-11:
[image]

dostt66, 2025-02-11:
denedim aynı hatayı aldım

dostt66, 2025-02-11:
[image]

fixajteknik, 2025-02-11:
bende bi deneyeim aynısını belki e22 güncellediyseler bizim pcb üzerinden parametre ayarı verilmiyor olabilir. bu sırada sende soketlerin yerini değişebilir misin bi aşağı yukarı kaydır dene.

dostt66, 2025-02-11:
denedim

fixajteknik, 2025-02-11:
hocam şu kodu dene sendeki kod eski pcb için software serial yanlış pinde tanımlı sendekinde

```
#include <SoftwareSerial.h>
 
SoftwareSerial fixSerial(3, 4); //PCB versiyon 4.3 den sonra bu şekilde olmalı
 
#define M0 7
#define M1 6
 
void setup() {
  Serial.begin(9600);
  fixSerial.begin(9600);
 
  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  digitalWrite(M0, LOW);        // E22 Modülleri için Aktif edin
  //digitalWrite(M0, HIGH);       // E32 Modülleri için Aktif edin
  digitalWrite(M1, HIGH);
}
 
void loop() {
  if (Serial.available()) {
    fixSerial.write(Serial.read());
  }
 
  if (fixSerial.available()) {
    Serial.write(fixSerial.read());
  }
}
```

fixajteknik, 2025-02-11:
[image]

birkaç defa denedim oldu bende ilkinde almadın get deyince

fixajteknik, 2025-02-11:
yinede tavisyemdinamik parametre ayarlarıdır. bu kodları inceleyin bu kodlar program başlamadan önce paramtre ayarlarını verir sonra işe başlar. youtube videosunda da kodun nasıl çalıştığını anlatıım.

VİDEO: https://www.youtube.com/watch?v=13nKOj_uG2s

KOD: https://github.com/fixajteknik/YouTube_Tutorials/tree/main/Video%20108%20%23define%20Part%202

closed 2025-08-14 by fixajteknik