fixajteknik/YouTube_Tutorials #26: ardinio nano ve e22 var birbirleri arası iletişim sağlayamadım

dostt66, 2025-02-12:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

ts

### Yorum yazdığınız kullanıcı adı nedir?

meraklı yazılımcı

### Arduino IDE veya IDF versiyonu.

ıde 22

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

1 Yıldan Az

### Eğitim seviyeniz

Üniversite

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

veri gönderiyorum alamıyorum iletişim sağlayamadım 

#include "LoRa_E22.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(3, 4);  // TX = 3, RX = 4
LoRa_E22 E22(&mySerial);

#define M0 7
#define M1 6

void setup() {
  Serial.begin(9600);
  mySerial.begin(9600);

  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  digitalWrite(M0, LOW);
  digitalWrite(M1, LOW);

  Serial.println("LoRa Gönderici Başladı...");
  E22.begin();

  // ⚙️ LoRa Modül Ayarlarını Yapılandır
  Configuration configuration;
  ResponseStructContainer c = E22.getConfiguration();
  configuration = *(Configuration*)c.data;

  configuration.ADDH = 0x00;
  configuration.ADDL = 0x01;  // Gönderici Adresi: 0x0001 (Alıcıdan farklı olmalı)
  configuration.CHAN = 18;    // Kanal: 18 (Alıcı ile aynı olmalı)

  ResponseStatus rs = E22.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
  Serial.print("Ayar Güncelleme Durumu: ");
  Serial.println(rs.getResponseDescription());
}

void loop() {
  String mesaj = "Merhaba, LoRa!";
  
  ResponseStatus rs = E22.sendMessage(mesaj);
  Serial.print("Gönderildi: ");
  Serial.println(mesaj);
  Serial.print("Gönderme Durumu: ");
  Serial.println(rs.getResponseDescription());

  delay(2000);  // 2 saniye bekle ve tekrar gönder
}

alıcı 
 #include "LoRa_E22.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(3, 4);  // TX = 3, RX = 4
LoRa_E22 E22(&mySerial);        // Hatalı satır düzeltildi!

#define M0 7
#define M1 6

void setup() {
  Serial.begin(9600);
  mySerial.begin(9600);

  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  digitalWrite(M0, LOW);
  digitalWrite(M1, LOW);

  Serial.println("LoRa Alıcı Başladı...");
  E22.begin();

  // ⚙️ LoRa Modül Ayarlarını Yapılandır
  Configuration configuration;
  ResponseStructContainer c = E22.getConfiguration();
  configuration = *(Configuration*)c.data;

  configuration.ADDH = 0x00;
  configuration.ADDL = 0x02;  // Alıcı Adresi: 0x0002 (Göndericiden farklı olmalı)
  configuration.CHAN = 18;    // Kanal: 18 (Gönderici ile aynı olmalı)

  ResponseStatus rs = E22.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
  Serial.print("Ayar Güncelleme Durumu: ");
  Serial.println(rs.getResponseDescription());
}

void loop() {
  if (E22.available() > 0) {
    ResponseContainer rs = E22.receiveMessage();
    Serial.print("Alındı: ");
    Serial.println(rs.data);
  } else {
    Serial.println("⏳ Beklenen veri yok...");
  }
  delay(1000);
}

 kodlarım bu

### Ama gerçekte ne oldu?

veri iletişimi sağlayamadım

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

1. Adım
2. Adım
3. Adım
...

### Log çıktıları varsa ekleyin.

```plain
iletişim sağlanmıyor kodlarıma baka bilirmisin
```

### Daha fazla bilgi

[image]
[image]
[image]
[image]

fixajteknik, 2025-02-12:
net id yi tanımlamamışsın. dün sana attığım kodu kullansan bu sorunları yaşamazsın

dostt66, 2025-02-12:
[image]

fixajteknik, 2025-02-12:
> [image]

burada adresleri ne yaptın kodları tamamını alıcı ve verici kod olarak ekler misin bakalım

dostt66, 2025-02-12:
alıcı kodu 
/*
** Öncelikle Hangi modül ile çalıştığınızı seçin e32 veya e22 arasında 
** Çalıştığınız modülün yanındaki "//" yoruma çek işaretini kaldırıp kullanmadığınız
** modülün başına "//" yazarak onu yoruma çekiniz. 
** https://shop.fixaj.com/
*/

#define e32433t20d        //e32 modülünü kullanıyorsanız burayı aktif edin
//#define e22900t22d      //e22 modülünü kullanıyorsanız burayı aktif edin

#ifdef e22900t22d
#include "LoRa_E22.h"
#define kutuphane LoRa_E22
#else
#include "LoRa_E32.h"
#define kutuphane LoRa_E32
#endif

////////////////////////////////     KART TANIMLAMALARI    /////////////////////////////////////////////////////////////
#if CONFIG_IDF_TARGET_ESP32  // ESP 32
#include <HardwareSerial.h>
#define ESP32AYAR
#define M0 32  //3in1 PCB mizde pin 7
#define M1 33  //3in1 PCB mizde pin 6
#define RX 27  //3in1 PCB mizde pin RX
#define TX 35  //3in1 PCB mizde pin TX

HardwareSerial fixSerial(1);  //Serial biri seçiyoruz.
kutuphane FixajSerial(TX, RX, &fixSerial, UART_BPS_RATE_9600);
//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#elif CONFIG_IDF_TARGET_ESP32S3  // S3
#include <HardwareSerial.h>
#define M0 4                    //  3in1 PCB mizde pin 7
#define M1 6                    //  3in1 PCB mizde pin 6
#define RX 18                   //  esp32 s3 de Lora RX e bağlı
#define TX 17                   //  esp32 s3 de Lora TX e bağlı

HardwareSerial fixSerial(1);
kutuphane FixajSerial(TX, RX, &fixSerial, UART_BPS_RATE_9600);
//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#elif defined ARDUINO_AVR_NANO  // NANO
#include <SoftwareSerial.h>
#define M0 7
#define M1 6
#define RX 3
#define TX 4
SoftwareSerial fixSerial(RX, TX);  //PCB versiyon 4.3 den sonra bu şekilde olmalı
kutuphane FixajSerial(&fixSerial);
//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//#error Hedef CONFIG_IDF_TARGET tanimlanmamis
#endif

#define Adres 2              //0--65000 arası bir değer girebilirsiniz. Diğer Modüllerden FARKLI olmalı
#define Kanal 20             //Frekans değeri Diğer Modüllerle AYNI olmalı.
#define GonderilecekAdres 1  //Mesajın gönderileceği LoRa nın adresi
#define Netid 63             //0--65000 arası bir değer girebilirsiniz. Diğer Modüllerle AYNI olmalı.

struct Signal {
  char sifre[15] = "Fixaj.com";
  char konum[15];
  bool btn1;
  byte btn2[10];
  byte btn3[4];
} data;

void setup() {
  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);

[... 320 lines cut ...]
  //configuration.SPED.airDataRate = AIR_DATA_RATE_000_03;  //Veri Gönderim Hızı 0,3 En uzak Mesafe
  //configuration.SPED.airDataRate = AIR_DATA_RATE_101_192; //Veri Gönderim Hızı 19,2 En Hızlı

  configuration.OPTION.transmissionPower = POWER_20;  //Geönderim Gücü max Varsayılan
  //configuration.OPTION.transmissionPower = POWER_10;  //Geönderim Gücü min
  //configuration.OPTION.transmissionPower = POWER_30; // E32 30d modülleri için bunu aktif et

  //GELİŞMİŞ AYARLAR
  configuration.SPED.uartBaudRate = UART_BPS_9600;
  configuration.SPED.uartParity = MODE_00_8N1;
  configuration.OPTION.fec = FEC_0_OFF;
  //configuration.OPTION.fec = FEC_1_ON;
  configuration.OPTION.fixedTransmission = FT_FIXED_TRANSMISSION;
  //configuration.OPTION.fixedTransmission = FT_TRANSPARENT_TRANSMISSION;
  configuration.OPTION.wirelessWakeupTime = WAKE_UP_250;
  configuration.OPTION.ioDriveMode = IO_D_MODE_PUSH_PULLS_PULL_UPS;

  // Ayarları KAYDET ve SAKLA
  ResponseStatus rs = FixajSerial.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
  Serial.println(rs.getResponseDescription());
  Serial.println(rs.code);
  c.close();
}

#endif

fixajteknik, 2025-02-15:
Simdi ilk once gorsellerde e22 kullandiginiz anlaşılıyor ama kodda e32 lora modulunu aktif etmissiniz neden

fixajteknik, 2025-02-15:
Kod eklerken <> eklediginiz kodu secip bu tusa basin okuma zorlaşıyor

closed 2025-08-14 by fixajteknik