fixajteknik/YouTube_Tutorials #29: lora e32 433 t30d Veri Alışverişi Sorunu

KatilChick, 2025-02-16:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

TS1701359 

### Yorum yazdığınız kullanıcı adı nedir?

-

### Arduino IDE veya IDF versiyonu.

Arduino IDE 2.3.4

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

Uzman

### Eğitim seviyeniz

Üniversite

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

2 LoRa'nın haberleşmesini istiyorduk.

### Ama gerçekte ne oldu?

Hiçbir şey olmadı. Veri alamadık.

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

Herhangi bir çıktı yok.

### Log çıktıları varsa ekleyin.

```plain
Herhangi bir çıktı yok.
```

### Daha fazla bilgi

2 yıllık LoRa'ları 2 ay önceye kadar kullanabiliyordum ve sorun çıkmıyordu. Yakın zamanda kullandığımda çalışmadığı için yenilerini sipariş ettim. Yeni LoRa'lar hiç çalışmıyordu.

fixajteknik, 2025-02-16:
Bariz temassizlik sorunu var pcb kullanman gerekir white boardda her zman sorun yasarsin. 

Ve yeni rf setting i dener misin

https://www.cdebyte.com/pdf-down.aspx?id=2602

ankaralii, 2025-03-19:
lora e32 433t30d modülleri için rf settings'in v4.0 versiyonu yerine başka bir versiyonunu kullanman gerek. 
https://www.cdebyte.com/products/E32-433T30D/4#Downloads
buradan bu versiyonunu deneyerek parametreleri alabilirsin. 30d ve 20d modelleri farklı, sen 20d için rf settingse bakıyorsun.