fixajteknik/YouTube_Tutorials #33: Full Set E70 433NW14S Esp32, Anten, 3in1 PCB, Kutu

BneLino, 2025-03-10:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

TS0603478

### Yorum yazdığınız kullanıcı adı nedir?

ProjectBayraktar

### Arduino IDE veya IDF versiyonu.

daha o aşamaya gelemedim 

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

SIFIR

### Eğitim seviyeniz

Lise

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

Devre kartının altında PC usb kullanmayın  yazıyordu ilk defa bir proje kullanacağımdan ötürü bilgisizim peki ben bilgisayara nasıl bağlıyacam yani yazılımını yüklerken bilgisayara nasıl bağlıyacam veya doğru anladım dimi bilgisayara takarsam bu sorun oluşturur peki ne yapmam gerek? ayrıca güç için esp32 den usb girişinden güç vermeliyim değilmi veya farklı bir yerden ise kaç v felan bilgi verirmisiniz?

### Ama gerçekte ne oldu?

Devre kartının altında PC usb kullanmayın  yazıyordu ilk defa bir proje kullanacağımdan ötürü bilgisizim peki ben bilgisayara nasıl bağlıyacam yani yazılımını yüklerken bilgisayara nasıl bağlıyacam veya doğru anladım dimi bilgisayara takarsam bu sorun oluşturur peki ne yapmam gerek? ayrıca güç için esp32 den usb girişinden güç vermeliyim değilmi veya farklı bir yerden ise kaç v felan bilgi verirmisiniz?

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

1. Adım
2. Adım
3. Adım
...

### Log çıktıları varsa ekleyin.

```plain

```

fixajteknik, 2025-03-11:
lipo pil ile beslerken bilgisayarın usb sinden beslenmeyin demek istiyordum orda.

BneLino, 2025-03-11:
anladım hocam şuanda iki tarafa da gerekli kodlar eklendi ama bağlantı testi kodları felan ekledim bağlantı zayıf hatalı diyor şimdi birşey deniyecem çözemez isem tekrardan sizden yardım isterim size bütün kodları mı atmalıyım çünkü espnin kodu 1.1K satır civarı da :D

BneLino, 2025-03-11:
kod doğru ama ayar moduna almıyor lorayı şuanda

BneLino, 2025-03-11:
Bu Node Dosyası

`#include <HardwareSerial.h>

// E70 LoRa modülü pin tanımlamaları
#define M0 32  // Mode pin 0
#define M1 33  // Mode pin 1
#define RX 27  // ESP32 RX - LoRa TX bağlantısı
#define TX 35  // ESP32 TX - LoRa RX bağlantısı

// UART haberleşme için HardwareSerial
HardwareSerial LoRaSerial(1); // UART1 kullanıyoruz

// Veri gönderme değişkenleri
int counter = 0;
unsigned long lastSendTime = 0;
const long sendInterval = 3000; // 3 saniyede bir veri gönder
bool ledState = false;

void setup() {
  // LED pin
  pinMode(LED_BUILTIN, OUTPUT);
  
  // Seri port başlat
  Serial.begin(115200);
  delay(1000);
  
  Serial.println("\n\n=== E70 LoRa Worker ===");
  
  // E70 Serial başlat (9600 baud)
  LoRaSerial.begin(9600, SERIAL_8N1, RX, TX);
  
  // M0 ve M1 pinlerini çıkış olarak ayarla
  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  
  // Modülü ayar moduna getir (AT modu)
  Serial.println("Modül AT moduna alınıyor...");
  digitalWrite(M0, HIGH);
  digitalWrite(M1, HIGH);
  delay(1000);
  
  // Seri port buffer'ı temizle
  while (LoRaSerial.available()) {
    LoRaSerial.read();
  }
  
  // AT modunda temel ayarları yap
  Serial.println("AT komutu gönderiliyor...");
  LoRaSerial.println("AT");
  delay(1000);
  showResponse();
  
  LoRaSerial.println("AT+VER");
  delay(500);
  showResponse();
  
  // Worker istasyon ayarları
  Serial.println("WMCFG ayarlanıyor...");
  LoRaSerial.println("AT+WMCFG=1"); // Node modu (1=Node)
  delay(500);
  showResponse();
  
  Serial.println("PIDCFG ayarlanıyor...");
  LoRaSerial.println("AT+PIDCFG=1234"); // Network ID - Merkez ile aynı olmalı!
  delay(500);
  showResponse();
  
  Serial.println("CNCFG ayarlanıyor...");
  LoRaSerial.println("AT+CNCFG=0"); // Kanal 0 - Merkez ile aynı olmalı!
[... 220 lines cut ...]
    // Manuel mesaj gönder
    if (command.startsWith("send ")) {
      String message = command.substring(5);
      LoRaSerial.println(message);
      Serial.println("Manuel mesaj gönderildi: " + message);
    }
    else if (command == "reset") {
      ESP.restart();
    }
  }
  
  delay(10);
}

// LoRa yanıtlarını ekrana yaz
void showResponse() {
  if (LoRaSerial.available()) {
    String response = LoRaSerial.readString();
    Serial.print("Yanıt: ");
    Serial.println(response);
  }
}
`

lorayı aldığımdan beri asla kullanmadım hata da vermiyor iletişim kurmuyorlar

BneLino, 2025-03-12:
Test için at moduna geçmesini sağlayan kod ekledim ama 
AT komutu gönderiliyor...
Yanıt alınamadı!
böyle bir dönüt aldım

fixajteknik, 2025-03-12:
Kodu duzgun yapistirilmadigi icin rahat okuyamiyorum ama sanirim format hatali.  Benim ornek kodu bi deneyin once

https://github.com/fixajteknik/YouTube_Tutorials/blob/main/Video%2089%20Derinlemesine%20E70/Coordinator_Merkez_e70_Genel_OK/Coordinator_Merkez_e70_Genel_OK.ino

BneLino, 2025-03-15:
ustam senin kodu denedim şehir içinde bir anten 4. kattaydı yanımdaki binalar ortalam 3 4 katlı sadece sokağın sonunda ortalama 200 metrede bağlantı koptu sebebini anlayamadım rssı uzaklaştıkca 160a kadar düştü yakında 200 civarlarındaydı en son birşey oldu anlamadım esp32 yandı normalde aldığım tek pil yetmiyordu iki pili seri bağlayıp takmıştım 8V civarında sistem uzun süredir düşük sıcaklıkta takılıyordu ama Lora modülünüzün güç ayarını 3. adıma aldım sanırım kodu gönderir göndermez esp32 bozuldu ve yandı sorun tam olarak neyden kaynaklandı çözemedim powersupyl olarak şuanda powerbank deneyeceğim tekrardan sipariş verdim sitenizden zaten bu E70 modülünüz dökümantasyon dosyasını nerden bulabilirim çünkü kodlama bilgim yok ve pahalı AI agentlar ile çalışıyorum onlara analiz ettirmem gerek E70 hakkında Ai sistemlerin bilgisi yetersiz pin girişlerini tanımlayamıyor

BneLino, 2025-03-17:
evet tekrar sipariş ettim gönderdiğiniz ürünler üstü pislik gönderdiğiniz pcb desen pinler yok üstü boş bu destek talebinde olduğu gibi telefonla aradım cevap vermediniz eski gönderilen neden yandı bilmiyorum bunlarda çalışmıyor lütfen artık geri dönüş yaparmısınız?

fixajteknik, 2025-03-17:
Her soru icin ayri baslik acmalisiniz ben sordugunuz soruya cevap verdikten sonra ayni baslik altinda yeni sorarsaniz gozden kacabiliyor.