fixajteknik/YouTube_Tutorials #34: Alacağım ürün hakkında emin değilim

Terzioglu10, 2025-03-11:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

almadım

### Yorum yazdığınız kullanıcı adı nedir?

Terzioglu10 youtube burak terzioğlu

### Arduino IDE veya IDF versiyonu.

ürün almadım

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

Hobi amaçlı

### Eğitim seviyeniz

Lise

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

https://www.hackster.io/mircemk/diy-extremly-sensitive-and-cheap-arduino-seismometer-50de60
https://www.hackster.io/mircemk/diy-extremly-sensitive-and-cheap-geophone-sensor-83187c
buradaki projeyi yapmak istiyorum bir de esp32 uyumlu mu diye merak ettim bir de modül lorawan rak almayı düşünüyorum. https://www.direnc.net/lm358-smd-single-supply-dual-operational-amplifiers https://www.direnc.net/1n4148-diyot-mic bu parçalar uyumlu mu? Uyumlu değilse arduino nano almayı düşünüyorum mağazanızdan. Her türlü alışveriş yapacağım ama ürün olarak ne alacağımdan emin değilim. İlk attığım link ucuza sismograf arduino projesi. Fakat ben almışken iyi bir cihaz alayım gerektiğinde başka işlerde kullanırım düşüncesiyle esp32 almayı düşünüyorum. Yardımcı olursanız sevinirim. Bir de ürünleri kararlaştırdığımda tek paket bir fiyat seçenekli sepete eklememi sağlayabilirseniz güzel olur.(olmasa da sıkıntı olmaz.) Devre becerim çok düşük. Yazılım bilgim chat gpt çalıştırıp kodları düzenleyebilme. Bilgisayar kullanma bilgim iyi. Bir de youtubeda gördüm yeni bir kart tasarlıyordunuz o kart mı bana daha uygun sizce?

### Ama gerçekte ne oldu?

Sismograf yapıp iletişime geçebilen bir elektronik proje yapmak. Mümkün değilse farklı bir proje yapacağım. Diğer devre elemanlarını mümkünse bu siteden değilse kendim temin edeceğim.

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

1. Adım
2. Adım
3. Adım
...

### Log çıktıları varsa ekleyin.

```plain

```

### Daha fazla bilgi

[image]
[image]
[image]

Terzioglu10, 2025-03-11:
https://www.komponentci.net/lm358-100x-sinyal-yukseltec-amfi-modulu-pmu6716?srsltid=AfmBOooMxAnQc9q7G9lpa3Ouqt_5t9eReEyW0ijkcFS6pktywCCyZ-gz yanlış yazdım doğrusu buymuş

fixajteknik, 2025-03-11:
GÜZEL SORU BİRAZ ARAŞTIRIP DÖNECEĞİM

fixajteknik, 2025-03-17:
Deneyebildiniz mi hocam

Terzioglu10, 2025-03-18:
> Deneyebildiniz mi hocam

Deneyemedim tam olarak nasıl bağlayacağımı çözemedim bir de lm358'i gelmesini bekliyorum https://www.youtube.com/watch?v=4NV-EX3gTtg bu video işime yarar mı

Terzioglu10, 2025-03-19:
> Deneyebildiniz mi hocam

bunu bağlayabilmem için  yardım edebilir misiniz benim üstte attığım görüntüyü yapabilmem için hangi kabloyu almalıyım mümkünse esp32 ile yapmayı deneyeceğim olmazsa ekstra olarak arduino alacağım. Arduinonun kodlarını chatgpt ile esp 32 uyumlu hale getirdim kod olarak sıkıntım olmayacak fakat devreyi nasıl bağlasam bilemedim.

fixajteknik, 2025-03-19:
bu flame sensörlerinin devresinden kopya edebilirsin. google flame sensor schematic yaz benzer devre çıkar