fixajteknik/YouTube_Tutorials #36: Lora reset hatası

Osmancidamal, 2025-04-08:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

TS2103525

### Yorum yazdığınız kullanıcı adı nedir?

Osman Çıdamal

### Arduino IDE veya IDF versiyonu.

Arduino ide 2.3.4

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

2 Yıldan Az

### Eğitim seviyeniz

Lise

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

[image]

[image]

[image]

[image]

https://github.com/user-attachments/assets/31715542-5986-4263-b947-d855290c8cbd

lora e22-900 t22 d elime ulaştı ve  belirttiğim şema ile çalışıyordum (arduino mega pro 2560) ve kod çalışıyordu. 1. olarak verici loranın bağlı olduğu aux ledi devre gücü kesip tekrar bağlayınca sönük yanmaya ve veri göndermemeye başladı. Bu sorunu arduinoya güç bağlıyken lorayı pinlerden çıkarıp takarak çözdüm ama bu sorunun projemde olmamsı lazım. bunun çözümünü arıyorum lorayı 3.3v ile besliyorum girişi ölçtüm. 2. sorun olarak verici loraya birkaç kez kod attıktan sonra lora ekndini resetliyor ve rf settings den tekrar ayar yapmam gerekiyor.

### Ama gerçekte ne oldu?

1. led düşük parlaklıkta yanıp veri göndermedi
2. lora parametre ayarları resetlendi

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

### Log çıktıları varsa ekleyin.

```plain

```

ankaralii, 2025-04-08:
GPS ve LoRa aynı anda UART'ı kullanan sensörler. Bu yüzden UART buffer hatası alıyorsun. Interrupt yöntemi ile çözebilirsin.

ankaralii, 2025-04-09:
parametre ayarını rf settings ile yapmadan https://fixaj.com/kolay-parametre-ayarlari/ linkindeki gibi yapabilirsin. burada alıcı ve vericinin parametre ayarlarına dokunmadan yap başta bi

fixajteknik, 2025-04-17:
Sorun çözüldü acaba

Osmancidamal, 2025-04-17:
hayır hocam malesef çözülmedi öneriniz var mı

fixajteknik, 2025-04-17:
Kodlari yukle bakarim

Osmancidamal, 2025-07-25:
hocam merhabalar bu projeye geri döndüm ve kodları gönderiyorum: 

[gpsbmp_verici-20250725T221223Z-1-001.zip](https://github.com/user-attachments/files/21438119/gpsbmp_verici-20250725T221223Z-1-001.zip)
[gpsbmp_alici-20250725T221222Z-1-001.zip](https://github.com/user-attachments/files/21438118/gpsbmp_alici-20250725T221222Z-1-001.zip)

Osmancidamal, 2025-08-14:
Abi bakar mısın lütfen

Osmancidamal, 2025-08-14:
Teknofest roket yarışması için acil