fixajteknik/YouTube_Tutorials #37: LoRa E22 Haberleştirme Sorunu

zyept, 2025-04-16:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

TS1404584

### Yorum yazdığınız kullanıcı adı nedir?

Zeynep Yiğit

### Arduino IDE veya IDF versiyonu.

Arduino IDE

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

Uzman

### Eğitim seviyeniz

Üniversite

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

2 Lora E22 900T22D ve 2 arduino mega 2560 kullandım. Vericinin Hello mesajını göndermesini ve alıcının bu mesajı almasını istiyordum. Alıcının adresi 1, alıcı ve verici aynı kanalda ->18). Bağlantıları aşağıda görebilirsiniz:

[image]

Yazdığım Kodlar:
Verici Kodu:
#include "LoRa_E22.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(17, 16); //RX,TX
LoRa_E22 E22(&mySerial);

void setup() {

  Serial.begin(9600);
  E22.begin();
  
  delay(500);
  Serial.println("Verici başladı");
}

void loop() {
  const char mesaj[] = "Hello";

  ResponseStatus rs = E22.sendFixedMessage(0, 1, 18, mesaj, sizeof(mesaj));
  
  Serial.print("Gönderildi: ");
  Serial.println(mesaj);
  Serial.println(rs.getResponseDescription());

  delay(2000); // her 2 saniyede bir mesaj gönder
}

Alıcı kodu: 
#include "LoRa_E22.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(17, 16);  // RX, TX
LoRa_E22 E22(&mySerial);

void setup() {
  Serial.begin(9600);
  E22.begin();
  delay(500);
  Serial.println("Alıcı hazır, mesaj bekleniyor...");
}

void loop() {
  if (E22.available() > 1) {
    ResponseContainer rc = E22.receiveMessage();

    if (rc.status.code == 1) {
      String gelenMesaj = rc.data;  // Artık doğrudan String

      Serial.print("Gelen mesaj: ");
      Serial.println(gelenMesaj);

      if (gelenMesaj == "Hello") {
        Serial.println("Hello mesajı başarıyla alındı!");
      }
    } else {
      Serial.print("Alım hatası: ");
      Serial.println(rc.status.getResponseDescription());
    }
  }
}

### Ama gerçekte ne oldu?

Vericiye kodu yükledim ve videoda da görebilirsiniz led yanıp sönüyor yavaş yavaş. Alıcıya kodu yüklediğimde ise alıcıdaki led yanıp sönmüyor sabit yanıyor. Hello mesajı gelmiyor.

https://github.com/user-attachments/assets/bc088800-4e9d-4edb-a233-41ebcf8f0414

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

Hata yok sadece alıcı Loraya mesaj gelmiyor.

### Log çıktıları varsa ekleyin.

```plain

```

### Daha fazla bilgi

Parametre ayarlarını yaptım. Alıcı adresi 1,kanal 18. Verici adresi 0,kanal 18. NET ID ve Şifre ayarlamadım

fixajteknik, 2025-04-17:
parametre ayarlarını görebilir miyim . fixed mod seçilimi

zyept, 2025-04-17:
[image] 

[image]
Tran mode normal yapmıştım

fixajteknik, 2025-04-17:
fixed yap, adres sıfır olur mu hiç denemedim, birini 1 diğerini 2 yap

zyept, 2025-04-17:
Alıcı adresini 2,verici adresini 1 yaptım. İkisi de fixed modda. Mavi led verici, yeşil led alıcı. Kod çıktılarını aşağıda paylaşıyorum:

https://github.com/user-attachments/assets/caa78a2f-3571-4c2e-820a-d699c1e3ba35
https://github.com/user-attachments/assets/8098f839-6ad0-430d-95bf-ee0d91e4e4ce

Alıcı çıktı:      12:49:49.844 -> Alıcı hazır, mesaj bekleniyor...
Verici çıktısı: 12:54:56.059 -> Gönderildi: Hello
                    12:54:56.059 -> Success

fixajteknik, 2025-04-17:
Bence software seriale gerek yok. Ayrica kodda mi yazdiginda mi bir tutarsizlik var sen message kisminda 1 numarali adrese mesaj atiyorsun ama yorumda alici adresi 2 demissin. Kafam karisti. Bizim ornek kodlar gibi struct ile yazsana

zyept, 2025-04-21:
Pardon yanlış yazmışım. Alıcı adresi 1, verici adresi 2. struct tanımlayarak kodu bu şekillde değiştirdim.
Alıcı kodu:
#include "LoRa_E22.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(17, 16);  // RX, TX
LoRa_E22 E22(&mySerial);

// Yapı tanımı (struct)
struct Mesaj {
  char metin[20];  // Gelen mesaj metni
};

void setup() {
  Serial.begin(9600);
  E22.begin();
  delay(500);
  Serial.println("Alıcı hazır, mesaj bekleniyor...");
}

void loop() {
  if (E22.available() > 1) {  // Mesaj varsa
    // Mesajı al
    ResponseStructContainer rsc = E22.receiveMessage(sizeof(Mesaj));

    if (rsc.status.code == 1) {  // Eğer mesaj başarıyla alındıysa
      struct Mesaj* gelenMesaj = (struct Mesaj*)rsc.data;  // Mesajı struct olarak al

      Serial.print("Gelen mesaj: ");
      Serial.println(gelenMesaj->metin);  // Gelen mesajı yazdır

      if (strcmp(gelenMesaj->metin, "Hello") == 0) {  // Mesaj "Hello" ise
        Serial.println("Hello mesajı başarıyla alındı!");
      }

      rsc.close();  // Veriyi serbest bırak
    } else {
      Serial.print("Alım hatası: ");
      Serial.println(rsc.status.getResponseDescription());  // Alım hatasını yazdır
    }
  }
}

Verici kodu:
#include "LoRa_E22.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(17, 16);  // RX, TX
LoRa_E22 E22(&mySerial);

// Yapı tanımı (struct)
struct Mesaj {
  char metin[20];  // Gönderilecek mesaj metni
};

void setup() {
  Serial.begin(9600);
  E22.begin();
  delay(500);
  Serial.println("Verici başladı");
}

void loop() {
  struct Mesaj mesaj;  // Mesaj yapısını oluştur
  strcpy(mesaj.metin, "Hello");  // Mesajı "Hello" olarak ata

  ResponseStatus rs = E22.sendFixedMessage(0, 1, 18, &mesaj, sizeof(mesaj));  // Mesaj gönder

  Serial.print("Gönderildi: ");
  Serial.println(mesaj.metin);  // Gönderilen mesajı seri port üzerinden yazdır
  Serial.println(rs.getResponseDescription());  // Gönderim durumunu yazdır

  delay(2000);  // her 2 saniyede bir mesaj gönder
}
 Ama hiçbir şey değişmedi. Alıcıya mesaj gelmiyor

fixajteknik, 2025-04-28:
Struct dedigim bu sekilde degildi. 

https://github.com/fixajteknik/YouTube_Tutorials/tree/main/Video%20108%20%23define%20Part%202/RSS%C4%B0%20li%20e22900t22

Bu kodu incele ve bu kodu kendi pcb ne gore uyarla mega da software serial olnaz hardware serial kullanmalisim. Chatgpt veya Claudia sor sana burdaki kodlari senin kullandigin çipe cevirir. Once benim kodu dene

zyept, 2025-05-05:
Dediğiniz gibi yaptım ama olmadı. Bu sefer bi arduino uno bi mega ile denedim. Alıcı mega:#include "LoRa_E22.h"

#define M0 7
#define M1 6
#define RX 17
#define TX 16

LoRa_E22 FixajSerial(TX, RX, &Serial2, UART_BPS_RATE_9600);

#define Adres 1
#define Kanal 18
#define GonderilecekAdres 2
#define Netid 63

struct Signal {
  char sifre[15];
  char konum[15];
  bool btn1;
  byte btn2[10];
  byte btn3[4];
} data;

void setup() {
  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  Serial.begin(9600);
  FixajSerial.begin();

  digitalWrite(M0, LOW);
  digitalWrite(M1, HIGH); // Ayar moduna geç
  delay(200);

  ResponseStructContainer c = FixajSerial.getConfiguration();
  Configuration configuration = *(Configuration*)c.data;
  configuration.ADDL = lowByte(Adres);
  configuration.ADDH = highByte(Adres);
  configuration.NETID = Netid;
  configuration.CHAN = Kanal;
  configuration.TRANSMISSION_MODE.fixedTransmission = FT_FIXED_TRANSMISSION;
  FixajSerial.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
  c.close();

  digitalWrite(M0, LOW);
  digitalWrite(M1, LOW); // Normal moda geç
  delay(200);

  Serial.println("Alıcı başlatıldı.");
}

void loop() {
  if (FixajSerial.available()) {
    ResponseStructContainer rsc = FixajSerial.receiveMessageRSSI(sizeof(Signal));
    Signal data = *(Signal*)rsc.data;

    if (strcmp(data.sifre, "Fixaj.com") == 0) {
      Serial.println("✅ Şifre doğru!");
      Serial.print("📍 Konum: "); Serial.println(data.konum);
      Serial.print("🔘 Buton durumu: "); Serial.println(data.btn1);
      Serial.print("📌 Koordinat: "); Serial.println(*(float*)(data.btn2), 6);
      Serial.print("🔢 Sayı: "); Serial.println(*(int*)(data.btn3));
      Serial.print("📶 RSSI: "); Serial.println(rsc.rssi);
    }

    rsc.close();
  }
}

Verici uno:#include <SoftwareSerial.h>
#include "LoRa_E22.h"

#define M0 7
#define M1 6
#define RX 3
#define TX 4

SoftwareSerial fixSerial(RX, TX);
LoRa_E22 FixajSerial(&fixSerial);

#define Adres 2
#define Kanal 18
#define GonderilecekAdres 1
#define Netid 63

struct Signal {
  char sifre[15] = "Fixaj.com";
  char konum[15];
  bool btn1;
  byte btn2[10]; // float saklamak için
  byte btn3[4];  // int saklamak için
} data;

void setup() {
  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  Serial.begin(9600);
  FixajSerial.begin();

  digitalWrite(M0, LOW);
  digitalWrite(M1, HIGH); // Ayar moduna geç
  delay(200);

  ResponseStructContainer c = FixajSerial.getConfiguration();
  Configuration configuration = *(Configuration*)c.data;
  configuration.ADDL = lowByte(Adres);
  configuration.ADDH = highByte(Adres);
  configuration.NETID = Netid;
  configuration.CHAN = Kanal;
  configuration.TRANSMISSION_MODE.fixedTransmission = FT_FIXED_TRANSMISSION;
  FixajSerial.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
  c.close();

  digitalWrite(M0, LOW);
  digitalWrite(M1, LOW); // Normal moda geç
  delay(200);

  Serial.println("Verici başlatıldı.");
}

void loop() {
  data.btn1 = true;
  *(float*)(data.btn2) = 37.7749;
  *(int*)(data.btn3) = 1453;
  strcpy(data.konum, "Istanbul");

  ResponseStatus rs = FixajSerial.sendFixedMessage(highByte(GonderilecekAdres), lowByte(GonderilecekAdres), Kanal, &data, sizeof(Signal));
  Serial.println(rs.getResponseDescription());
  delay(2000);
}

zyept, 2025-05-06:
Merhaba tekrardan, iki tane arduino uno ile deneyince haberleştirdim. STM32 NUCLEO F756ZG ve Arduino Unoyu haberleştirmem gerekiyor. Alıcı arduino uno, verici stm32 nucleo f756zg. Haberleştirmeyi yapamadım STM32de kütüphane yok nasıl yaparım?Kullandığım kodları atıyorum. STM32CubeIDE KULLANDIM stm İÇİN

Alıcı:

#include <SoftwareSerial.h>
#include "LoRa_E22.h"

#define M0 7
#define M1 6
#define RX 3
#define TX 4

SoftwareSerial fixSerial(RX, TX);
LoRa_E22 FixajSerial(&fixSerial);

#define Adres 1
#define Kanal 18
#define GonderilecekAdres 2
#define Netid 63

struct Signal {
  char sifre[15];
  char konum[15];
  bool btn1;
  byte btn2[10];
  byte btn3[4];
} data;

void setup() {
  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  Serial.begin(9600);
  FixajSerial.begin();

  digitalWrite(M0, LOW);
  digitalWrite(M1, HIGH);
  delay(200);

  ResponseStructContainer c = FixajSerial.getConfiguration();
  Configuration configuration = *(Configuration*)c.data;
  configuration.ADDL = lowByte(Adres);
  configuration.ADDH = highByte(Adres);
  configuration.NETID = Netid;
  configuration.CHAN = Kanal;
  configuration.TRANSMISSION_MODE.fixedTransmission = FT_FIXED_TRANSMISSION;
  FixajSerial.setConfiguration(configuration, WRITE_CFG_PWR_DWN_SAVE);
  c.close();

  digitalWrite(M0, LOW);
  digitalWrite(M1, LOW);
  delay(200);

  Serial.println("Alıcı başlatıldı.");
}

void loop() {
  if (FixajSerial.available()) {
    ResponseStructContainer rsc = FixajSerial.receiveMessageRSSI(sizeof(Signal));
    Signal data = *(Signal*)rsc.data;

    if (strcmp(data.sifre, "Fixaj.com") == 0) {
      Serial.println("✅ Şifre doğru!");
      Serial.print("📍 Konum: "); Serial.println(data.konum);
      Serial.print("🔘 Buton durumu: "); Serial.println(data.btn1);
      Serial.print("📌 Koordinat: "); Serial.println(*(float*)(data.btn2), 6);
      Serial.print("🔢 Sayı: "); Serial.println(*(int*)(data.btn3));
      Serial.print("📶 RSSI: "); Serial.println(rsc.rssi);
    }

    rsc.close();
[... 278 lines cut ...]
  /* USER CODE BEGIN Error_Handler_Debug */
  /* User can add his own implementation to report the HAL error return state */
  __disable_irq();
  while (1)
  {
  }
  /* USER CODE END Error_Handler_Debug */
}

#ifdef  USE_FULL_ASSERT
/**
  * @brief  Reports the name of the source file and the source line number
  *         where the assert_param error has occurred.
  * @param  file: pointer to the source file name
  * @param  line: assert_param error line source number
  * @retval None
  */
void assert_failed(uint8_t *file, uint32_t line)
{
  /* USER CODE BEGIN 6 */
  /* User can add his own implementation to report the file name and line number,
     ex: printf("Wrong parameters value: file %s on line %d\r\n", file, line) */
  /* USER CODE END 6 */
}
#endif