fixajteknik/YouTube_Tutorials #40: LoRa modülü 24 saatte bir donuyor

otomasyon45, 2025-05-22:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

TS54634163

### Yorum yazdığınız kullanıcı adı nedir?

otomasyon45

### Arduino IDE veya IDF versiyonu.

Arduino IDE 2.3.4

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

2 Yıldan Az

### Eğitim seviyeniz

Üniversite

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

2 LoRa modülünü haberleştirmek istedim. 1 tanesi su tankının boşaldığını haber veriyor. Diğeri bu sinyali alıyor ve su pompasını çalıştırıyor. Sistem bu şekilde çalışıyor ama 24 saat sonra kitleniyor. Bağlantı kablolarını çıkarıp taktığımda düzeliyor. Ama 24 saat sonunda tekrar oluyor.

### Ama gerçekte ne oldu?

PANO---------------------------------------------------------------------------------
 #include "LoRa_E22.h"
#include <SoftwareSerial.h>
#include <avr/wdt.h>

SoftwareSerial mySerial(3, 4); // Arduino RX <-- e32 TX, Arduino TX --> e32 RX
LoRa_E22 E22(&mySerial);
 
String sifre = "";
#define M0 7
#define M1 6
#define StartBtn 8
#define Role2 9
#define LORA_RESET 5 // LoRa modülü reset pini

unsigned long previousMillis = 0;
unsigned long lastReceivedTime = 0; // Son mesaj alınan zaman
const unsigned long timeoutPeriod = 13000; // 13 saniye zaman aşımı

struct Signal {
  char type[15];
  bool role;
} data;
 
void setup() {
  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  pinMode(StartBtn, OUTPUT);
  pinMode(Role2, OUTPUT);
  pinMode(LORA_RESET, OUTPUT);
  
  digitalWrite(M0, LOW);
  digitalWrite(M1, LOW);
  digitalWrite(StartBtn, LOW);
  digitalWrite(Role2, LOW);
  digitalWrite(LORA_RESET, HIGH);

  Serial.begin(9600);
  E22.begin();
  delay(500);

  lastReceivedTime = millis(); // Başlangıç zamanı kaydet
  wdt_enable(WDTO_4S); // 4 saniyede bir reset olmazsa Arduino kendini resetler
}
 
void loop() {
  wdt_reset(); // Arduino'nun kilitlenmesini önler

  // 13 saniye boyunca mesaj gelmezse röleyi kapat
  if (millis() - lastReceivedTime >= timeoutPeriod) {
    Serial.println("13 saniye boyunca mesaj alınmadı, röle kapatılıyor.");
    digitalWrite(StartBtn, LOW);
    digitalWrite(Role2, LOW);
  }

  while (E22.available() > 1) {
    // Gelen mesaj okunuyor
    ResponseStructContainer rsc = E22.receiveMessage(sizeof(Signal));
    struct Signal data = (Signal) rsc.data;
    Serial.print("Gelen Mesaj: ");
    Serial.println(data.type);
   
    if (strcmp(data.type, "Şifre") == 0) { // Gelen şifreyi kontrol ediyor
      Serial.println("şifre doğru");
      if (data.role) { // Eğer butona basılıysa röleyi startlıyor
        Serial.println("role start");
        digitalWrite(StartBtn, HIGH); // Röle startlandı
        digitalWrite(Role2, HIGH); // 2.röle startlandı
      } else { // Eğer buton basılmıyorsa röleyi kapatıyor
        Serial.println("role stop");
        digitalWrite(StartBtn, LOW); // Röle kapandı
        digitalWrite(Role2, LOW); // 2.röle kapandı
      }
      lastReceivedTime = millis(); // Son mesaj alınma zamanını güncelle
    }

    rsc.close();
  }
}
 KUMANDA--------------------------------------------------------------------------------------------------
 #include "LoRa_E22.h"
#include <SoftwareSerial.h>
#include <avr/wdt.h>

SoftwareSerial mySerial(3, 4);
LoRa_E22 E22(&mySerial);

unsigned long previousMillis = 0;
unsigned long sendAgain = 0; // 10 saniye kontrolü için zamanlayıcı değişken

struct Signal {
  char type[15] = "Şifre";
  bool role;
} data;

#define M0 7
#define M1 6
#define BTN 2

bool isSending = false;  // Buton için kontrol
unsigned long buttonPressStart = 0; // Buton basılma başlangıç zamanı
bool buttonPressed = false; // Butonun basılı olup olmadığını kontrol

void setup() {
  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  digitalWrite(M0, LOW);
  digitalWrite(M1, LOW);

  pinMode(BTN, INPUT_PULLUP); // Buton

  Serial.begin(9600);
  E22.begin();
  delay(500);
}

void loop() {
  int buttonState = digitalRead(BTN); // Buton durumu oku

  if (buttonState == LOW) { // Buton basılı olduğunda
    if (!buttonPressed) { // İlk kez basıldıysa
      buttonPressed = true;
      buttonPressStart = millis(); // Basılma başlangıç zamanını kaydet
    }
    delay(500);

    // 2 saniye kontrolü
    if (!isSending && (millis() - buttonPressStart >= 2000)) {
      isSending = true; // Veri gönderme moduna geç
      data.role = 1;
      Serial.println("Buton 2 saniyeden fazla basılı, veri gönderiliyor!");
      Serial.print("data.role: ");
      Serial.println(data.role);
      ResponseStatus rs = E22.sendFixedMessage(0, 44, 25, &data, sizeof(Signal)); // Veriyi gönderdi
      Serial.println(rs.getResponseDescription());
      sendAgain = millis(); // 10 saniye zamanlayıcısını başlat
      delay(1500);

    } else if (isSending && (millis() - sendAgain >= 10000)) {
      // 10 saniye kontrolü
      data.role = 1;
      Serial.println("10 saniyede bir veri tekrar gönderiliyor");
      Serial.print("data.role: ");
      Serial.println(data.role);
      ResponseStatus rs = E22.sendFixedMessage(0, 44, 25, &data, sizeof(Signal)); // Veriyi gönderdi
      Serial.println(rs.getResponseDescription());
      sendAgain = millis(); // Zamanlayıcıyı sıfırla
      delay(1500);
    }
  } else { // Buton bırakıldığında
    if (buttonPressed) {
      buttonPressed = false;
      if (isSending) {
        data.role = 0;
        isSending = false; // Veri gönderme durumu kapat
        Serial.println("Buton bırakıldı, veri gönderimi durduruldu.");
        ResponseStatus rs = E22.sendFixedMessage(0, 44, 25, &data, sizeof(Signal)); // Veriyi gönderdi
        Serial.println(rs.getResponseDescription());
        delay(1500);
      } else {
        Serial.println("Buton 2 saniyeden kısa süre basıldı, işlem yapılmadı.");
      }
    }
  }

  delay(500); // Debounce için kısa gecikme
} 

Bu kod ile işlem gerçekleşiyor ama 24 saat sonunda sistem kitleniyor (Kumanda tarafı)

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

1. Adım
2. Adım
3. Adım
...

### Log çıktıları varsa ekleyin.

```plain

```

otomasyon45, 2025-05-22:
Ek olarak gece saat 3'de gidiyor ve saat 6'da hiç bişey olmamış gibi çalışmaya devam ediyor. Her gün neredeyse aynı saatlerde oluyor.

fixajteknik, 2025-05-22:
TS54634163 BÖYLE BİR SİPARİŞ NO YOK SİSTEMDE

closed 2025-08-15 by fixajteknik