fixajteknik/YouTube_Tutorials #43: ESP32S3 Dev Module seçiminde serial monitöre veri yazdıramama sorunu.

evscope, 2025-06-12:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

TS1405697

### Yorum yazdığınız kullanıcı adı nedir?

xx

### Arduino IDE veya IDF versiyonu.

2.3.6 cli version 1.2.0

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

Hobi amaçlı

### Eğitim seviyeniz

None

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

ESP32S3 Dev Module seçiminde serial monitöre veri yazdıramama sorunu.

### Ama gerçekte ne oldu?

void setup() {
  Serial.begin(115200);

  Serial.println("START");
}

void loop() {
  Serial.println("TEST");
  delay(1000);  // 1 saniye bekle
}
 bu kodda bile seriale birşey yazılmıyor

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

1. Adım
2. Adım
3. Adım
...

### Log çıktıları varsa ekleyin.

```plain

```

fixajteknik, 2025-06-12:
her güncelemede parametre ayarları sıfırlanıyor. O yüzden serial monitöre yazmasını istiyorsanız "**USB CDC on Boot**" ayarını enable etmelisiniz.

[image]

evscope, 2025-06-12:
[image]