fixajteknik/YouTube_Tutorials #48: S3 Röle Bağlantı Şemaları Hakkında

muratsuzen, 2025-07-23:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

TS

### Yorum yazdığınız kullanıcı adı nedir?

msuzen

### Arduino IDE veya IDF versiyonu.

2.2

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

SIFIR

### Eğitim seviyeniz

None

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

Merhaba ESP32 S3 PCB ye röle takmak istiyoruz. Röle bağlantı şemasını, resimlerini paylaşırmısınız.
[image]

### Ama gerçekte ne oldu?

Merhaba ESP32 S3 PCB ye röle takmak istiyoruz. Röle bağlantı şemasını, resimlerini paylaşırmısınız.
[image]

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

1. Adım
2. Adım
3. Adım
...

### Log çıktıları varsa ekleyin.

```plain

```

fixajteknik, 2025-07-23:
Sipariş no doğru yazar misinizGalaxy cihazımdan gönderildi
-------- Orijinal mesaj --------Kimden: Murat Süzen ***@***.***> Tarih: 23.07.2025  10:52  (GMT+03:00) Alıcı: fixajteknik/YouTube_Tutorials ***@***.***> Cc: Subscribed ***@***.***> Konu: [fixajteknik/YouTube_Tutorials] S3 Röle Bağlantı Şemaları Hakkında (Issue #48) muratsuzen created an issue (fixajteknik/YouTube_Tutorials#48)
Şart, TÜM CheckBox'ları işaretleyin.

 Videoları ***@***.***) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
 İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
 Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
 Satın aldığım ürünlere yorum yaptım.
 Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

Sipariş Numaranızı Yazınız
TS
Yorum yazdığınız kullanıcı adı nedir?
msuzen
Arduino IDE veya IDF versiyonu.
2.2
İşletim Sisteminiz
Windows
Yazılım bilginiz
SIFIR
Eğitim seviyeniz
None
Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.
Merhaba ESP32 S3 PCB ye röle takmak istiyoruz. Röle bağlantı şemasını, resimlerini paylaşırmısınız.
image.png (view on web)
Ama gerçekte ne oldu?
Merhaba ESP32 S3 PCB ye röle takmak istiyoruz. Röle bağlantı şemasını, resimlerini paylaşırmısınız.
image.png (view on web)
OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

Adım
Adım
Adım
...

Log çıktıları varsa ekleyin.

Daha fazla bilgi
No response

fixajteknik, 2025-07-23:
İlk önce burada role için  hangi pin ve kullanılan malzemelerin ürün linkleri tablo halinde verilmiş.  ve PCB üstündeki designator ile urunleri lehimleyince roleyi aktif edebilirsinizSchmatic ise şu an tatildeyim dönünce 2 haftaya hatırlat yine atarım. Galaxy cihazımdan gönderildi
-------- Orijinal mesaj --------Kimden: Murat Süzen ***@***.***> Tarih: 23.07.2025  12:40  (GMT+03:00) Alıcı: fixajteknik/YouTube_Tutorials ***@***.***> Cc: Mehmet YILDIZ ***@***.***>, Comment ***@***.***> Konu: Re: [fixajteknik/YouTube_Tutorials] S3 Röle Bağlantı Şemaları Hakkında (Issue #48) muratsuzen left a comment (fixajteknik/YouTube_Tutorials#48)
TS1405697

closed 2025-07-24 by muratsuzen