fixajteknik/YouTube_Tutorials #52: 3in1 ESP ve E22 900T30D kullanımı

sfkozclk, 2025-08-09:
### Şart, TÜM CheckBox'ları işaretleyin.

- [x] Videoları (https://www.youtube.com/@Fixaj) ve sitenizdeki örnek kodlara (https://fixaj.com/son-yazilarimiz/) baktım ama çözüm bulamadım.
- [x] İndirme sayfasındaki (https://fixaj.com/indirme-sayfasi/) yazılımları ve driverları yükledim ama çözüm bulamadım.
- [x] Ürünleri Fixaj Mağazalarından aldım. Sadece kendi sattığınız ürünlere destek veriyoruz.
- [x] Satın aldığım ürünlere yorum yaptım.
- [x] Soruya kullandığım kodları ve devre şema fotoğraflarını ekleyeceğim.

### Sipariş Numaranızı Yazınız

TS2507864

### Yorum yazdığınız kullanıcı adı nedir?

sfkozclk

### Arduino IDE veya IDF versiyonu.

Arduino IDE 2.3.6

### İşletim Sisteminiz

Windows

### Yazılım bilginiz

Hobi amaçlı

### Eğitim seviyeniz

Üniversite

### Ne olmasını istiyordunuz ama olmadı? Projenizin özeti ve devre şeması ekleyebilirsiniz.

3in1 ESP PCB ile Lora E22 900T30D'yi çalıştırmak istiyorum.

### Ama gerçekte ne oldu?

3in1 ESP PCB'nin üzerinde Lora için 30D, 20D ve 22D için ayrı pin noktaları tanımlanmış ve bunlardan yalnızca 20D ve 22D üzerine socket lehimlenmiş. Benim satınaldığım 900T30D modülünün bağlanacağını düşündüğüm 30D yazan 7 adet pin'de socket lehimli değil. User manuallerde yaptığım araştırmada 22D ve 30D'nin bacak bağlantılarının aynı olduğunu gördüm. Hal böyle olunca 30D ve 22D 'nin aynı socket'e takılabilmesi gerekir ancak 30D için ayrı bir pin alanı yapılmış.

Sizin youtube'da yer alan 4 adet lora 900t30D'yi haberleştirdiğiniz videoda seçebildiğim kadarıyla bunları PCB üzerinde 22d de yer alan socket girişine takılı olduğunu gördüm. 

Sonuç olarak 900T30D'yi PCB deki 30D yazan kısma socket lehimleyip buraya mı takmalıyız, yoksa PCB üzerindeki 22D ya da 20D olan socketlerden birine takarak da çalıştırabilir miyiz?

### OUTPUT çıktısı nedir? Hatanın oluşma anını yazar mısınız?

1. Adım
2. Adım
3. Adım
...

### Log çıktıları varsa ekleyin.

```plain

```

fixajteknik, 2025-08-11:
BAHSETTİĞİNİZ GİBİ AYNI PİN YAPISINDA OLDUĞU İÇİN İSTEDİĞİNİZ YERE TAKIP ÇALIŞTIRABİLİRSİNİZ SORUN OLMAZ. 30D İÇİN PCB ÜSTÜNDEKİ SOKET YERİNİN AMACI 30D MODÜLÜ BİRAZ DAHA UZUN OLDUĞU İÇİN KUTUYA RAHAT SIĞSIN DİYE AZ BERİYE ÇEKTİK. YOKSA TAMAMEN AYNI.

**EK OLARAK** : BAZEN ARKADAŞLAR PİLLE VS BESLEME DURUMLARINDA GERİLİM ÇÖKMEMESİ İÇİNCE DE BOŞ SOKETİN VCC VE GND ARASINA KONDANSATOR EKLEDİKLERİ OLUYOR. GÜVEMLİK AMAÇLI.

sfkozclk, 2025-08-15:
Teşekkürler